luxonis/depthai-ros-examples #10: fatal error No such file or directory. C++ header files cannot be found.

canberkgurel, 2021-08-24:
I'm having the `No such file or directory` error when I try to build my catkin_ws. C++ header files cannot be found.

`Errors     << depthai_examples:make /home/ubuntu/catkin_ws/logs/depthai_examples/build.make.002.log
/home/ubuntu/catkin_ws/src/depthai-ros-examples/depthai_examples/lib/rgb_pipeline.cpp:1:10: fatal error: depthai_examples/rgb_pipeline.hpp: No such file or directory
 #include <depthai_examples/rgb_pipeline.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/depthai_examples.dir/lib/rgb_pipeline.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/ubuntu/catkin_ws/src/depthai-ros-examples/depthai_examples/lib/rgb_stereo_pipeline.cpp:1:10: fatal error: depthai_examples/rgb_stereo_pipeline.hpp: No such file or directory
 #include <depthai_examples/rgb_stereo_pipeline.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/ubuntu/catkin_ws/src/depthai-ros-examples/depthai_examples/lib/stereo_from_host.cpp:1:10: fatal error: depthai_examples/stereo_from_host.hpp: No such file or directory
 #include <depthai_examples/stereo_from_host.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/depthai_examples.dir/lib/rgb_stereo_pipeline.cpp.o] Error 1
make[2]: *** [CMakeFiles/depthai_examples.dir/lib/stereo_from_host.cpp.o] Error 1
make[1]: *** [CMakeFiles/depthai_examples.dir/all] Error 2
make: *** [all] Error 2
`

I tried sourcing ` /opt/ros/melodic/setup.bash` but that didn't solve it.

saching13, 2021-08-24:
Hey Canberk,
    I spun a new Ubuntu 18.04 docker and tried the complete steps, but I was not able to reach this error. 
Can you try deleting the `devel` and `build` followed by updating the repository and giving it a shot again. 
if it doesn't work can you provide the steps you carried out and on which system so that I can recreate this.

canberkgurel, 2021-08-24:
@saching13 Let's compare notes, what steps have you taken?

saching13, 2021-08-24:
I spun up `ros-base` melodic docker and just followed all the steps [here](https://github.com/luxonis/depthai-ros) without a single modification.