luxonis/depthai-ros-examples #13: Noisy Pointcloud after launching the stereo_node.launch

hardesh-peppermint, 2021-11-12:
Hi,

I just got a new OAK D 2 days ago. I've been tinkering with the camera. I noticed that the camera's giving very noisy pointcloud. I could find a solution online. I've uploaded a [bag file](https://drive.google.com/drive/folders/1eBkMXG3zH-D0YMHjwWTZ89fTIv0ybdNo?usp=sharing) in case you want to take a look at the pointcloud and the images. 

Kindly look into this. 

Thanks,
Harshal

hardesh-peppermint, 2021-11-12:
Hi,

Joined the discord channel after posting this issue. I'll be trying out the lastest commit on depthai-core. Will let you know if the performance improves.

saching13, 2021-11-14:
Is this ROS1 or ROS2 ?.

hardesh-peppermint, 2021-11-15:
I'm using ROS1. 

Performance improved significantly after installing the latest version of depthai-core. There's still some noise, but I think that can be removed by filtering.

saching13, 2021-11-16:
Some noise is still there since it is raw depth. We are working on some post processing which will be included in upcoming releases

hardesh-peppermint, 2021-11-16:
Thanks for letting me know. Looking forward towards newer releases 😄 

We can close this issue if you want.

closed 2021-11-16 by saching13