luxonis/depthai-ros-examples #22: /depthai-ros-examples/depthai_examples/ros1_src/stereo_nodelet.cpp bug!

tosa-no-onchan, 2022-02-22:
Hi!
Would you update some code?

/depthai-ros-examples/depthai_examples/ros1_src/stereo_nodelet.cpp
...
line 148
    std::tuple<dai::Pipeline, int, int> createPipeline(bool withDepth, bool lrcheck, bool extended, bool subpixel, int confidence, int LRchecktresh, std::string resolution){
        dai::Pipeline pipeline;
        ...
        // MonoCamera
        //monoLeft->setResolution(dai::MonoCameraProperties::SensorResolution::THE_720_P);
        monoLeft->setResolution(monoResolution);    // here
        monoLeft->setBoardSocket(dai::CameraBoardSocket::LEFT);
        //monoRight->setResolution(dai::MonoCameraProperties::SensorResolution::THE_720_P);
        monoRight->setResolution(monoResolution);   //  here
        monoRight->setBoardSocket(dai::CameraBoardSocket::RIGHT);
        ...

saching13, 2022-02-22:
Sorry for the issue.
cc: @daxoft

saching13, 2022-02-22:
Fixed it.

tosa-no-onchan, 2022-02-22:
Thank you for your early response,seaching13.
It worked!

I am testing with OAK-D-Lite.
On PC Ubuntu Mate 18.04 and ROS melodic.

closed 2022-03-22 by saching13