luxonis/depthai-ros-examples #32: I want to subscribe to the topic "color / yolov4_Spatial_detections" and get the xyz coordinates from "geometry_msgs / Point position"

yo4hi6o, 2022-04-04:
I'm writing a program that subscribes to the topic "color / yolov4_Spatial_detections", gets the xyz coordinate value from "geometry_msgs / Point position", gets it with MOVEIT, and does pick and place work.

For now, I'm creating a simple node that subscribes to xyz from "yolov4_Spatial_detections".

``

saching13, 2022-04-04:
> void clbk(const depthai_ros_msgs::SpatialDetection::ConstPtr& msg) {
You are using the wrong message. 
You are subscribing to `SpatialDetectionArray` and passing it to `SpatialDetection` Message here.

yo4hi6o, 2022-04-05:
thank you.

I modified the code as below. 

```
#include <ros/ros.h>
#include "geometry_msgs/Point.h"
#include <depthai_ros_msgs/SpatialDetection.h>
#include <depthai_ros_msgs/SpatialDetectionArray.h>

//void clbk(const depthai_ros_msgs::SpatialDetection::ConstPtr& msg) {
void clbk(const depthai_ros_msgs::SpatialDetectionArray::ConstPtr& msg) {

ROS_INFO("position: x=%.2f, y=%.2f", msg->detections[0].position.x, msg->detections[0].position.y);
   
}

int main(int argc, char **argv)
{
  // ROS objects
  ros::init(argc, argv, "yolov4_subscriber_node");
  ros::NodeHandle n;
  ros::Subscriber sub = n.subscribe("/yolov4_publisher/color/yolov4_Spatial_detections", 1, clbk);

  ros::spin();

}

```
The previous error is gone. However, when I start it, the subscribe node goes down.

Startup procedure
```
roslaunch depthai_examples yolov4_publisher.launch   //T1
 roslaunch depthai_examples yolov4_subscriber.launch  //T2
```
master.log
```
[rosmaster.master][INFO] 2022-04-05 09:39:16,569: publisherUpdate[/yolov4_publisher/color/yolov4_Spatial_detections] -> []: sec=0.00, exception=[Errno 111] Connection refused
[rosmaster.threadpool][ERROR] 2022-04-05 09:39:16,570: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1095, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 898, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 860, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 837, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused

```

yo4hi6o, 2022-04-12:
@saching13 
Thank you for your support. A sample subscribe node would be helpful. Thank you.

yo4hi6o, 2022-04-17:
@saching13
Thankyou.
I didn't know how to subscribe to an array message file.
 I was able to solve it with reference to [this issue](https://answers.ros.org/question/60614/how-to-publish-a-vector-of-unknown-length-of-structs-in-ros/). Thank you.

```
[ INFO] [1650183310.524148357]: 
 position:
 x=0.030027
 y=-0.099319
 z=1.019924
```

closed 2022-04-17 by yo4hi6o