luxonis/depthai-ros-examples #35: Why not merge this repository into depthai-ros?

windelbouwman, 2022-04-11:
It might be good to merge this repository into the depthai-ros repository, the repositories are related anyway, and are a cohesive set of ros packages. At this moment readme instructions refer across repositories, and you need both anyway. Users not wanting the examples can use colcon to exclude the depthai_ros_examples package.

The folder structure in the depthai-ros repository could then become:

- `README.md` with instructions on usage
- `depthai_ros_msgs` as is
- `depthai_bridge` as is
- `depthai_examples` merged from depthai-ros-examples repo

This would ease the use of the ros packages a bit.

Luxonis-Brandon, 2022-04-12:
Thanks!  Asking internally.

danilo-pejovic, 2023-05-23:
We will use this repository from now on to add some use cases for our ROS nodes and packages mentioned here are already transferred to depthai-ros. 

As issue here is resolved, I will be closing this issue.

closed 2023-05-23 by danilo-pejovic