luxonis/depthai-ros-examples #36: Weird disparity values being published in stereo_node

psFournier, 2022-04-13:
Hello,
I am trying to retrieve a disparity image from the stereo_node publisher with mode != depth, and the values observed, be it from a subscriber to .../stereo/disparity or with a rostopic echo .../stereo/disparity, seem off: 
most are 0 with some integer values scattered everywhere. I checked that the disparity image obtained directly from the device with a getFrame call does not produce these values but "proper values" ranging between 1 and 255.

Also, I noticed with rostopic echo that the delta_d attribute of the published DisparityImage is 0.0, whereas the source code for the DisparityConverter seems to produce values that are either 1 or 1/32.

Could these two observations be related ?
Thanks in advance

saching13, 2022-04-14:
Sorry for the issue. We are looking into it.

daxoft, 2022-04-14:
Hi, can you try to turn subpixel off? As for the delta error, it was a cast bug, it will be merge into main soon, if you need it fixed now you can try this branch for the depthai-ros repo: https://github.com/luxonis/depthai-ros/tree/daxoft/fix_disparity

saching13, 2022-04-14:
@psFournier  Can you check this viewer to verify ? the image looks good to me, 
` rosrun image_view disparity_view image:=/stereo_publisher/stereo/disparity`

saching13, 2022-04-14:
[image]

FPSychotic, 2022-04-14:
Same me I think, but I didn't make test. Just black disparity with
normalised ticked or not. ROS2 warp branch

closed 2023-05-23 by danilo-pejovic