luxonis/depthai-ros-examples #37: OAK-D Pro W Dev  ROS and ROS2 support add basic options equally to every example

FPSychotic, 2022-04-17:
Please give support to OAK-D Pro W Dev  in all 
ROS/ROS2 examples .
I received the camera 3 weeks ago but I only was able so stream the stereo topics without depth (no depth data but published).

Another different problem is different examples have different options, even the basic ones that should be common in all of them as resolution, rectification, etc.

I can see the camera advertised in "ROS" discurse while it cannot be considered supported IMHO.

Any timeline to add it?

saching13, 2022-04-17:
Hello,
Use the branch in the PR to install the depthai-core. That will fix the depth problem considering your device is calibrated. No update req for depthai-ros

https://github.com/luxonis/depthai-core/pull/437

FPSychotic, 2022-04-17:
That is the PR I have installed. It only give me 2 stereo streams and the depth is published, but very weird. Maybe is just bad visualization in rviz but anyways It don't allow me use stereo_inertial or anything with rgb camera as rgb stereo example. I need the IMU and rgb and that is not working.
For other side If I connect OAK-D  instead the Pro W to have color, it makes artifacts due to a bug in the opencv mesh function. 
Is what resulted from  the Discord but nothing changed it looks.
No sure if luxonis can reproduce the error, what is a bug and what is job that needs be done by luxonis, but right now in my case, with the things advided in Discord, from I got the camera three weeks ago, ROS is not working, and now OAK-D is bugged too

FPSychotic, 2022-04-17:
https://youtu.be/3ci0qoSXKbc
This is the problem I get with OAK-D and that branch. In Pro-W I think worked better, but blurred corners

saching13, 2022-04-17:
Ah ok. Sorry. looks like I missed the conv on discord. Let me test it and get back. 

In the meantime can you check if you see the same artifacts on [this](https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/stereo_depth_video.py) python example in main  with `-lm` argument ?

FPSychotic, 2022-04-17:
I tested both cameras with -lm and without it. 
-PRO W Dev , With -lm gets depth in a reasonable pattern, in ROS2 doesnt , depth has a lot less noise than without -lm, , better framerate as without -lm gets laggy. corners still blured in both.
[image]
[image]
[image]
[image]
now the right sensors pictures

[image]
[image]

-OAK-D, now the artifacts are not, no even the desaligment one I showed in discord, but in the depth, I can se clearly, that both erros, desaligment and distorsions, just like the depth is made from the unrectified images. 

Can you check to be sure the depth in both cameras is done from the rectified images? in the OAK-D case Im quite sure at least.

saching13, 2022-04-18:
[image]

[image]

This is my OAK-D-W. and it is using the `warp_mesh_on_device`. I don't see any issues with using it on this branch with depthai-ros

On the artifacts in OAK-D. it is coming from OpenCV calibration. I'm looking into it.

FPSychotic, 2022-04-18:
I'm in ROS2 is that ROS2?. Is it inertial-stereo node? , Is it working rgb examples?.

saching13, 2022-04-18:
> I'm in ROS2 is that ROS2?.

Yes it is ros2. 

> Is it inertial-stereo example? 

First image is not. Second image is. 

>  Are working the rgb examples?.

What is this ? We don't have rectification added for RGB. we will have to add in the FW or add an ImageManip node to do that.

saching13, 2022-04-18:
Everything that works on ROS2 will work on ROS1.

FPSychotic, 2022-04-18:
So rgb exmples not working?

For what add something that leave a sensor off of game?

Could be that any panameter in my inertial stereo example is calling a RGB function and because that stereo inertial crash?

If not, Im afraid that I received another camera without calibration

saching13, 2022-04-18:
All dev versions of devices are not factory calibrated. 

> So rgb exmples not working?

I'm not understanding what you mean. Can you share which RGB example is not working so that I can check ?

> Could be that any panameter in my inertial stereo example is calling a RGB function and because that stereo inertial crash?

Stil not clear. Please share the command used. And the error.

FPSychotic, 2022-04-18:
"What is this ? We don't have rectification added for RGB. we will have to add in the FW or add an ImageManip node to do that."

Any rgb/spatial example that use rgb camera.
Before change to mesh system, was rectified the RGB in camera or in host? How could be  colororized the pointclouds then?

FPSychotic, 2022-04-18:
[image]

FPSychotic, 2022-04-18:
Alex and Nejc said it is calibrated

FPSychotic, 2022-04-19:
If I run stereo_inertial_node 
`igcs@igcs:~$ ros2 launch depthai_examples stereo_inertial_node.launch.py
[INFO] [launch]: All log files can be found below /home/igcs/.ros/log/2022-04-19-07-55-42-890342-igcs-4655
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [stereo_inertial_node-1]: process started with pid [4670]
[INFO] [robot_state_publisher-2]: process started with pid [4672]
[INFO] [component_container-3]: process started with pid [4674]
[INFO] [component_container-4]: process started with pid [4676]
[INFO] [rviz2-5]: process started with pid [4678]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link oak-d_frame had 5 children
[robot_state_publisher-2] Link oak_imu_frame had 0 children
[robot_state_publisher-2] Link oak_left_camera_frame had 1 children
[robot_state_publisher-2] Link oak_left_camera_optical_frame had 0 children
[robot_state_publisher-2] Link oak_model_origin had 0 children
[robot_state_publisher-2] Link oak_rgb_camera_frame had 1 children
[robot_state_publisher-2] Link oak_rgb_camera_optical_frame had 0 children
[robot_state_publisher-2] Link oak_right_camera_frame had 1 children
[robot_state_publisher-2] Link oak_right_camera_optical_frame had 0 children
[robot_state_publisher-2] [INFO] [1650351343.116838981] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-2] [INFO] [1650351343.116903378] [oak_state_publisher]: got segment oak-d_frame
[robot_state_publisher-2] [INFO] [1650351343.116913994] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-2] [INFO] [1650351343.116923843] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-2] [INFO] [1650351343.116934389] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-2] [INFO] [1650351343.116944028] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-2] [INFO] [1650351343.116953457] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-2] [INFO] [1650351343.116963445] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-2] [INFO] [1650351343.116974062] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-2] [INFO] [1650351343.116984049] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-3] [INFO] [1650351343.339841596] [container]: Load Library: /opt/ros/foxy/lib/libdepth_image_proc.so
[component_container-4] [INFO] [1650351343.341705210] [container2]: Load Library: /opt/ros/foxy/lib/libdepth_image_proc.so
[component_container-3] [INFO] [1650351343.361529355] [container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-3] [INFO] [1650351343.362063044] [container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-4] [INFO] [1650351343.362286619] [container2]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-4] [INFO] [1650351343.362318748] [container2]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-4] [INFO] [1650351343.362331180] [container2]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-4] [INFO] [1650351343.362342635] [container2]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-4] [INFO] [1650351343.362353042] [container2]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-4] [INFO] [1650351343.362365055] [container2]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-4] [INFO] [1650351343.362376440] [container2]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-3] /opt/ros/foxy/lib/rclcpp_components/component_container: symbol lookup error: /opt/ros/foxy/lib/libdepth_image_proc.so: undefined symbol: _ZN15image_transport19create_subscriptionEPN6rclcpp4NodeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFvRKSt10shared_ptrIKN11sensor_msgs3msg6Image_ISaIvEEEEEESA_17rmw_qos_profile_t
[component_container-4] /opt/ros/foxy/lib/rclcpp_components/component_container: symbol lookup error: /opt/ros/foxy/lib/libdepth_image_proc.so: undefined symbol: _ZN15image_transport19create_subscriptionEPN6rclcpp4NodeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFvRKSt10shared_ptrIKN11sensor_msgs3msg6Image_ISaIvEEEEEESA_17rmw_qos_profile_t
[ERROR] [component_container-3]: process has died [pid 4674, exit code 127, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=container -r __ns:=/'].
[ERROR] [component_container-4]: process has died [pid 4676, exit code 127, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=container2 -r __ns:=/'].
[rviz2-5] [INFO] [1650351343.643232265] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1650351343.643346811] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1650351343.687398783] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1650351343.799307499] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] Parsing robot urdf xml string.
[... cut ...]
[ERROR] [rgb_stereo_node-1]: process has died [pid 4863, exit code -6, cmd '/home/igcs/ros2_depthai_ws/install/depthai_examples/lib/depthai_examples/rgb_stereo_node --ros-args --params-file /tmp/launch_params_rpe6y3k5 --params-file /tmp/launch_params_hld3jbzb --params-file /tmp/launch_params_ho_y_4jl --params-file /tmp/launch_params_bfzxex29 --params-file /tmp/launch_params_u3u972s6 --params-file /tmp/launch_params_toggfmy5 --params-file /tmp/launch_params_0r2lcpix --params-file /tmp/launch_params_pr7fi6ux --params-file /tmp/launch_params_hpoj7hpp --params-file /tmp/launch_params_2u43061y --params-file /tmp/launch_params_hxxjdosq --params-file /tmp/launch_params__sbswrzs'].
`

FPSychotic, 2022-04-19:
If I run yolo4 whatever, different error

` ros2 launch depthai_examples yolov4_publisher.launch.py
[INFO] [launch]: All log files can be found below /home/igcs/.ros/log/2022-04-19-08-02-29-666155-igcs-5021
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)
`

It was  a clean installation.

saching13, 2022-04-19:
> [stereo_inertial_node-1] what(): There is no Camera data available corresponding to the the requested cameraId

Your camera is not calibrated. I checked with Nejc. 
he said he didn't calibrate any Dev versions.

FPSychotic, 2022-04-19:
> > [stereo_inertial_node-1] what(): There is no Camera data available corresponding to the the requested cameraId
> 
> Your camera is not calibrated. I checked with Nejc. he said he didn't calibrate any Dev versions.
> 
> Sorry for any miscommunications that might have happened on Discord

I understood by this message it would be calibrated.
I will post the issues from now here, as it looks that in Discord there is issues. I will try recover the time, 

"You will get a board with OV9782 WFOV color camera and OV9282 WFOV mono cameras already installed on the board. So the cameras work out of box already. But active illumination is not supported yet, as they share an I2C bus with the cameras'.

By the way, ask please support the illuminator, it is what make them PRO XD.

And just confirm, what branch should I use to calibrate? Do you have any picture I can put in a 42" tv  to calibrate, is the only thing I have.

saching13, 2022-04-19:
> In Discord we checked with a python script and worked, Alex I think was, said if it ls not calibrated, wont work, and worked, so we assumed it was calibrated.

When he says it won't work. it means you won't get valid depth. In ros we need to publish the Camera Info so Calibration is needed here.  Sorry for the miscommunication. 

> By the way, ask please support the illuminator, it is what make them PRO XD.

Will add this to the Stereo inertial node today. 

> And just confirm, what branch should I use to calibrate? Do you have any picture I can put in a 42" tv to calibrate, is the only thing I have.

you can use main if this is not a custom version of OAK-D Pro W. 
Does it has 2 OV9282 and 1 OV9782 ?

FPSychotic, 2022-04-19:
Yes , it is a standard Pro W, probably as the project use 3D lidar and make interference I will add two color stereo modules and the IR in the center. But not now, maybe I add notch filter or buy filtered IR sendor.But would be nice have support to different sensor combinations in ROS, anyways I guess the one that have like rpi modules I guess need that too

saching13, 2022-04-19:
if it's a standard one you should be able to calibrate on main branch of `depthai` repo

FPSychotic, 2022-04-19:
> if it's a standard one you should be able to calibrate on main branch of `depthai` repo
Sorry , only for calibration, or I just can install main with the pro W

FPSychotic, 2022-04-19:
Untill now I need Warp branch

saching13, 2022-04-19:
that you did on C++ right. Not for python. If for python also it should be fine I think but better to follow the steps in the link below

[https://docs.luxonis.com/en/latest/pages/calibration/](https://docs.luxonis.com/en/latest/pages/calibration/)

FPSychotic, 2022-04-27:
> you did on C++ right. Not for python

That doesn't work.
I cannot use the camera in ROS because it is not calibrated.
I need calibrate it to have the rgb recognised
I need the rgb recognized to can calibrate.
The fov in the .json file doesn't match with adviced fov either.
So what I do next.

[image]

saching13, 2022-04-27:
replied on discord. 
need to use [this](this) branch for OAK-D-Pro-W

closed 2022-06-03 by FPSychotic