luxonis/depthai-ros-examples #38: OAK-D Factory sensor misalignment affects depth

FPSychotic, 2022-04-17:
My new OAK-D came with the sensors misaligned. In discord I was said it is normal.
After tests and two different branch, with mesh and without mesh in the calibration, the depth in both is affected.
Even if I aim the camera to an area which the most near area is 3m, the misaligned areas, gets distance of 0 or close to 0, and in the best case much closer than 3 meters, it appears red, and as it has valor (have no depth would be much better), it will prevent my drone or rovers of be armed or will be stuck or will have an undetermined behaviour as they will detect an obstacle in front of them, and in the better case they will be replanning continuosly.

Just it cannot be used, after show this video I would like luxonis confirm it is normal.  Maybe ROS devs understand better the issue that is generating and how will affect to a robot 

https://youtu.be/HyfL8lXYvUo

saching13, 2022-04-18:
> Even if I aim the camera to an area which the most near area is 3m, the misaligned areas, get a distance of 0 or close to 0.

This means that there is no Feature or at the corners it his getting hard to detect. At the very edge of the image it is often makes this noise. But  is the depth quality affected in the center and other places too ? 
The video is hard to get what you are trying to explain. I'm only seeing that it is not in detail at the edges,

saching13, 2022-04-18:
That said on ROS side we are working on our own post-processing steps so in that stage will try to clear those noisy points.

FPSychotic, 2022-04-19:
> This means that there is no Feature or at the corners it his getting hard to detect. At the very edge of the image it is often makes this noise. But is the depth quality affected in the center and [...]
I try explain me better.

In the video I sequentially I change between rectified streams. 
-How you consider the misalignment? as normal? 
-have you seen ever 2 rectified images in a stereo camera on which there is an uncoupled  rotation? both images dont fit in the vertical because a sensor is so rotated that cannot be rectified. There is a further effect here that is in ROS the rectified images have the black band and distortion and can be seen the effect in the depth. In python effect can be seen in the depth, but the black band doesn't appear, but still the picture is rotated and the effect seen in the depth. So you there are seeing the better, in ROS it is still worse.(maybe by the warp mesh branch?)
-Do you think that such thing will affect to the disparity? I left a example in Discord, so yes will affect I guess.
How much affects? OAK-D depth quality always was bad or very bad, how I can measure that? Can you? I never seen a good quality in a OAK-D.
-So the point is, is the right sensor rotated or not? in a rectified image should be rotated or not? by the physics law and how disparity is calculated, that will affect to the depth or not? 
In the video the top right red band continuously is giving wrong measurement, no idea if red means 0 or means 10cm or means no disparity, but will affect that to a robot in perception or not.
Have you develop a ROI or something to can remove that wrong areas of the image?.

closed 2022-06-03 by FPSychotic