luxonis/depthai-ros-examples #39: SpatialLocationCalculator distortion with setDepthAlign()

smellslikeml, 2022-04-20:
Hi team!
I want to visualize detections with Markers on a Depth Cloud using the [depthai_example yolov4_publisher](https://github.com/luxonis/depthai-ros-examples/blob/main/depthai_examples/ros1_src/yolov4_spatial_publisher.cpp) so I configured the stereo node with `setDepthAlign(dai::CameraBoardSocker::RGB)`. 

Although we can visualize the depth cloud, this setting appears to distort the spatial location calculation. Specifically, most estimates are near (x=-2, y=0, z=4) or undefined (0, 0, 0).  [see control video vs variant video attached].

Investigating further, I tried the `dai::StereoDepthConfig::AlgorithmControl::DepthAlign::CENTER` enum found [here](https://github.com/luxonis/depthai-shared/blob/main/include/depthai-shared/datatype/RawStereoDepthConfig.hpp#L25), which appears to restore location calculation. However, the resolutions of the image frame and the depth frame do not align using this enumerator so we have no depth cloud. Thank you for your attention.

[discord discussion](https://discord.com/channels/790680891252932659/962104883807387708)

https://user-images.githubusercontent.com/9044907/164233057-9baeaa43-7c41-417c-bf3f-dbc15682cd7e.mp4

https://user-images.githubusercontent.com/9044907/164233140-6ea3f101-c57d-4978-a6e2-4bd529e22c4b.mp4

saching13, 2022-04-20:
Can you share what plugin is used here to visualize the detected object as marker so that we can recreate it on our side ?

smellslikeml, 2022-04-23:
Hi @saching13, thanks for looking into this!

For visualization, I created a simple marker publisher node similar to this [gist](https://gist.github.com/smellslikeml/1626f20d3e68f87f6c990b052f06d6ed). To rule out visualization issues, I computed spatial locations in a script node from monocular depth estimates shown in the video attached.

https://user-images.githubusercontent.com/9044907/164910702-67dd337d-6906-4c2b-beee-6f4e6034947c.mp4

Independent of visualization, I was able to see problematic spatial coordinates results from echo'ing the rostopic.

danilo-pejovic, 2023-05-23:
Hey, sorry for random ping year later. We are repurposing this repo to host ROS examples we develop over time (which are not critical to regular use of our ROS nodes). Let me know if this issue is still relevant and in that case we can move it to depthai-ros.

closed 2023-05-23 by danilo-pejovic