luxonis/depthai-ros-examples #46: devel branch dont work

FPSychotic, 2022-05-16:
I rebuild it following the detpthai-ros , 20.04,noetic from apt, arm64, but main branch works
sometimes crash , sometimes just do nothing . made it work partially removing things, but 400p, IMU, detections, marker dont work 

 roslaunch depthai_examples stereo_inertial_node.launch 
... logging to /home/ubuntu/.ros/log/22af78e4-d535-11ec-9bbf-db787525b9cb/roslaunch-ubuntu-15634.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.170:36775/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://192.168.43.170:11311/

process[oak_state_publisher-1]: started with pid [15674]
process[stereo_inertial_publisher-2]: started with pid [15675]
process[nodelet_manager-3]: started with pid [15676]
process[depth_image_convertion_nodelet-4]: started with pid [15681]
process[depth_image_to_rgb_pointcloud-5]: started with pid [15683]
process[rviz-6]: started with pid [15689]
[ INFO] [1652718489.992307036]: Initializing nodelet with 4 worker threads.
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
Listing available devices...
Device Mx ID: 14442C10F1B6C2D200
[14442C10F1B6C2D200] [880.336] [SpatialDetectionNetwork(8)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
Device USB status: SUPER
[ WARN] [1652718499.589680484]: Accel data not found. Dropping data
^C[rviz-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[... 151 lines cut ...]
        138:       if (joinable())
      > 139: 	std::terminate();
        140:     }
        141: 
        142:     thread(const thread&) = delete;
#4    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa640727f, in std::terminate()
#3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa640721b, in 
#2    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa640983b, in __gnu_cxx::__verbose_terminate_handler()
#1    Source "/build/glibc-RIFKjK/glibc-2.31/stdlib/abort.c", line 79, in abort [0xffffa61f1aab]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0xffffa6204d78]
Aborted (Signal sent by tkill() 16064 1000)
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 16064, exit code -6, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/22af78e4-d535-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2.log].
log file: /home/ubuntu/.ros/log/22af78e4-d535-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

saching13, 2022-05-16:
@FPSychotic  
Can you run it with `DEPTHAI_LEVEL=debug` and share the log please ?
@daxoft  can you help with this please ?

FPSychotic, 2022-05-16:
ok, I deleted manually and trying again, I think it could be finding a previous depthai version

FPSychotic, 2022-05-16:
No booted, but previous did too sometimes.
if I choose 400p and NN marker activated crash

`
process[depth_image_to_rgb_pointcloud-5]: started with pid [35312]
process[markerPublisher-6]: started with pid [35314]
process[rviz-7]: started with pid [35324]
[ INFO] [1652723099.698223002]: Initializing nodelet with 4 worker threads.
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
Listing available devices...
Device Mx ID: 14442C10F1B6C2D200
terminate called after throwing an instance of 'dai::XLinkWriteError'
terminate called recursively
  what():  Couldn't write data to stream: '__stream_asset_storage' (X_LINK_ERROR)
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 35304, exit code -6, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2.log].
log file: /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[markerPublisher-6] killing on exit
[nodelet_manager-3] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

FPSychotic, 2022-05-16:
I added it as argument not sure if that is ok

roslaunch depthai_examples stereo_inertial_node.launch arg_name:=DEPTHAI_LEVEL=debug

```
roslaunch depthai_examples stereo_inertial_node.launch arg_name:=DEPTHAI_LEVEL=debug
... logging to /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/roslaunch-ubuntu-37537.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.170:38999/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 400p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    markerPublisher (depthai_examples/markerPublisher.py)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://192.168.43.170:11311/

process[oak_state_publisher-1]: started with pid [37559]
process[stereo_inertial_publisher-2]: started with pid [37560]
process[nodelet_manager-3]: started with pid [37561]
process[depth_image_convertion_nodelet-4]: started with pid [37566]
process[depth_image_to_rgb_pointcloud-5]: started with pid [37568]
process[markerPublisher-6]: started with pid [37570]
process[rviz-7]: started with pid [37576]
[ INFO] [1652724417.364288008]: Initializing nodelet with 4 worker threads.
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
Listing available devices...
Device Mx ID: 14442C10F1B6C2D200
[14442C10F1B6C2D200] [1082.283] [SpatialDetectionNetwork(8)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[... 76 lines cut ...]
#6    Source "/tmp/depthai-core/src/device/DeviceBase.cpp", line 409, in tryStartPipeline [0xffff93c96a07]
#5    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff934fa5df, in __cxa_rethrow
#4    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff934fa27f, in std::terminate()
#3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff934fa21b, in 
#2    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff934fc8cb, in __gnu_cxx::__verbose_terminate_handler()
#1    Source "/build/glibc-RIFKjK/glibc-2.31/stdlib/abort.c", line 79, in abort [0xffff932e4aab]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0xffff932f7d78]
Aborted (Signal sent by tkill() 37560 1000)
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 37560, exit code -6, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2.log].
log file: /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[markerPublisher-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[oak_state_publisher-1] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

saching13, 2022-05-16:
`DEPTHAI_LEVEL=debug roslaunch depthai_examples stereo_inertial_node.launch `

It's an environment variable. Can you try this.

FPSychotic, 2022-05-16:
`DEPTHAI_LEVEL=debug roslaunch depthai_examples stereo_inertial_node.launch 
... logging to /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/roslaunch-ubuntu-38469.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.170:43699/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 400p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    markerPublisher (depthai_examples/markerPublisher.py)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://192.168.43.170:11311/

process[oak_state_publisher-1]: started with pid [38494]
process[stereo_inertial_publisher-2]: started with pid [38495]
process[nodelet_manager-3]: started with pid [38496]
process[depth_image_convertion_nodelet-4]: started with pid [38503]
process[depth_image_to_rgb_pointcloud-5]: started with pid [38507]
process[markerPublisher-6]: started with pid [38514]
process[rviz-7]: started with pid [38516]
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
[ INFO] [1652724890.589602489]: Initializing nodelet with 4 worker threads.
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
[2022-05-16 18:14:50.682] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-16 17:01:35 +0000
[2022-05-16 18:14:50.688] [debug] Initialize - finished
[2022-05-16 18:14:50.968] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1.tar.xz' open: 4ms, archive read: 262ms
[2022-05-16 18:14:52.079] [debug] Resources - Archive 'depthai-device-fwp-db9c1703c929df0de6cc99b198d2d149f1735598.tar.xz' open: 33ms, archive read: 1345ms
Listing available devices...
Device Mx ID: 14442C10F1B6C2D200
[2022-05-16 18:14:53.671] [debug] Device - OpenVINO version: 2021.4
[2022-05-16 18:14:53.673] [debug] Device - BoardConfig: {"gpio":[],"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
[... 106 lines cut ...]
#7    Source "/tmp/depthai-core/src/device/Device.cpp", line 27, in Device [0xffffa08fe06f]
#6    Source "/tmp/depthai-core/src/device/DeviceBase.cpp", line 409, in tryStartPipeline [0xffffa0910a07]
#5    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa01745df, in __cxa_rethrow
#4    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa017427f, in std::terminate()
#3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa017421b, in 
#2    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffffa01768cb, in __gnu_cxx::__verbose_terminate_handler()
#1    Source "/build/glibc-RIFKjK/glibc-2.31/stdlib/abort.c", line 79, in abort [0xffff9ff5eaab]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0xffff9ff71d78]
Aborted (Signal sent by tkill() 38495 1000)
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 38495, exit code -6, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2.log].
log file: /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[markerPublisher-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

saching13, 2022-05-16:
Looks like a power issue. Provide external power supply to the OAK or disable by setting `enableSpatialDetection=False` as argument to ROS launch file. That might fix it.

I willl update the example to disable it by default.

FPSychotic, 2022-05-16:
if leave as default with 720p and marker as false

`DEPTHAI_LEVEL=debug roslaunch depthai_examples stereo_inertial_node.launch 
... logging to /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/roslaunch-ubuntu-39071.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.170:40949/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://192.168.43.170:11311/

process[oak_state_publisher-1]: started with pid [39095]
process[stereo_inertial_publisher-2]: started with pid [39096]
process[nodelet_manager-3]: started with pid [39097]
process[depth_image_convertion_nodelet-4]: started with pid [39103]
process[depth_image_to_rgb_pointcloud-5]: started with pid [39105]
process[rviz-6]: started with pid [39106]
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
[ INFO] [1652725175.903029833]: Initializing nodelet with 4 worker threads.
 nn path.../home/ubuntu/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/tiny-yolo-v4_openvino_2021.2_6shave.blob
[2022-05-16 18:19:35.917] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-16 17:01:35 +0000
[2022-05-16 18:19:35.920] [debug] Initialize - finished
[2022-05-16 18:19:36.197] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1.tar.xz' open: 3ms, archive read: 269ms
[2022-05-16 18:19:37.045] [debug] Resources - Archive 'depthai-device-fwp-db9c1703c929df0de6cc99b198d2d149f1735598.tar.xz' open: 9ms, archive read: 1096ms
Listing available devices...
Device Mx ID: 14442C10F1B6C2D200
[2022-05-16 18:19:38.479] [debug] Device - OpenVINO version: 2021.4
[2022-05-16 18:19:38.481] [debug] Device - BoardConfig: {"gpio":[],"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
[... 55 lines cut ...]
[2022-05-16 18:19:42.361] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
#6    Source "/tmp/depthai-core/src/device/DataQueue.cpp", line 121, in setMaxSize [0xffff963db33b]
#5    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff95b62573, in __cxa_throw
#4    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff95b6227f, in std::terminate()
#3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff95b6221b, in 
#2    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff95b648cb, in __gnu_cxx::__verbose_terminate_handler()
#1    Source "/build/glibc-RIFKjK/glibc-2.31/stdlib/abort.c", line 79, in abort [0xffff9594caab]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0xffff9595fd78]
Aborted (Signal sent by tkill() 39096 1000)
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 39096, exit code -6, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2.log].
log file: /home/ubuntu/.ros/log/838b0b56-d53f-11ec-9bbf-db787525b9cb/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

`

FPSychotic, 2022-05-16:
[depth_image_convertion_nodelet-4-stdout.log](https://github.com/luxonis/depthai-ros-examples/files/8702487/depth_image_convertion_nodelet-4-stdout.log)
[depth_image_to_rgb_pointcloud-5-stdout.log](https://github.com/luxonis/depthai-ros-examples/files/8702488/depth_image_to_rgb_pointcloud-5-stdout.log)
[master.log](https://github.com/luxonis/depthai-ros-examples/files/8702489/master.log)
[roslaunch-ubuntu-39528.log](https://github.com/luxonis/depthai-ros-examples/files/8702490/roslaunch-ubuntu-39528.log)
[roslaunch-ubuntu-39579.log](https://github.com/luxonis/depthai-ros-examples/files/8702491/roslaunch-ubuntu-39579.log)
[rosout.log](https://github.com/luxonis/depthai-ros-examples/files/8702492/rosout.log)
[rosout-1-stdout.log](https://github.com/luxonis/depthai-ros-examples/files/8702493/rosout-1-stdout.log)
[rviz-6-stdout.log](https://github.com/luxonis/depthai-ros-examples/files/8702494/rviz-6-stdout.log)

FPSychotic, 2022-05-16:
> Looks like a power issue. Provide external power supply to the OAK or disable by setting `enableSpatialDetection=False` as argument to ROS launch file. That might fix it.
> 
> I willl update the example to disable it by default.

It is externally powered directly from a 5Amp 5.2v BEC connected to a LIPO 150A battery. If it has a power issue it is not from the power supply maybe from the internal components.

saching13, 2022-05-16:
Which device are you using ?

FPSychotic, 2022-05-16:
Tested with a wall adpater first brand and powering Jetson Nano and Aaeon X86 SBC without issues. And it is happening too

OAK-D

FPSychotic, 2022-05-16:
it crashed in both OAK-D, with different power supply, good ones, different cables, the original, but others too, from realsenses and one certified. Twice didn't crash in both cameras, but it wasn't showing the detection in the preview, so I assume it worked because the NN crashed in the background . After worked twice in both cameras, just started to crash again.

I guess it is easy to reproduce, cannot be my two cameras are no good and every cable a power supply.

FPSychotic, 2022-05-16:
I did a another test in a second rpi4 same problem, so two oaks , two RPi4, different cables and power supplies.
what I noted in the second rpi didnt send the IMU error, no idea, that is a different issue .

FPSychotic, 2022-05-16:
rgb_stereo example, the Rgb image is very laggy , in the stereo inertial when aligned is fine, Im not interested in this example but let you know.

FPSychotic, 2022-05-16:
The good news it looks it is working the auto config for Warp-dewarp mesh and it is working with homography I guess, I don't have black bands and the point cloud looks came back to the normality , noisy , but much better that with the mesh activated. Good it was rolled back, thanks by that.

FPSychotic, 2022-05-16:
With depth_aligned:=false nothing works , at 720p or 400p, no mono or depth image, but nothing crash, just no image.

I finish the tests here I hope you find something useful. When it works I would say it is a step forward if we leave the IMU aside.
cheers. Spatial is a very good thing to have.

FPSychotic, 2022-05-17:
OK, more tests in the end depth_aligned:=false in some specific setup worked and I can see the stereo streams, but it looks, stereo_inertial is still applying the mesh (bad thing) and the auto mode (auto mesh for fisheye) is not working but stereo_node is not applying the mesh, (good) what means or it wasn't changed by long time and it is not affected with the recent mesh changes, or it is taking the auto mode. So lucking this we are stepping back not forward.
[image]
[image]
[image]
the one good is the homography one stereo mode. the other two mesh in left and right

FPSychotic, 2022-05-17:
In other try is even more weird, it was without image in the right sensor by 1  minute, and after that appeared the image without the mesh applied, so Ok , but the other sensor as the mesh applied, so bad, as one have the the mesh and other not, there is no depth. The camera go in its own.

Also I show the IMU problem that affects to devel and Main branch , you can see how the imu is moving itself, it happens in both OAK-D

[https://youtu.be/jjtVldqyEhU](url)

FPSychotic, 2022-05-25:
Same thing in main branch.

saching13, 2022-05-27:
Duplicate of one in depthai-ros here
https://github.com/luxonis/depthai-ros/issues/113

closed 2022-05-27 by saching13