luxonis/depthai-ros-examples #49: unable to run yolo, mobilenet in Jetson NX

iam-shanmukha, 2022-05-25:
unable to run yolo, mobilenet in Jetson NX. Getting killed with error.`Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f9ffc04f8, in raise
Aborted (Signal sent by tkill() 30198 1000)`

Host Device: Nvidia Jetson NX
OAK Model: OAK-D-POE

#### Whats working
- can run depthai_demo.py
- can run `roslaunch depthai_examples stereo_inertial_node.launch`

#### Unable to run
- roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-POE
- roslaunch depthai_examples mobile_publisher.launch 

```
roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-POE
... logging to /home/nvidia/.ros/log/aa7cf384-dbf0-11ec-8a96-48b02d2f7086/roslaunch-forOak-30154.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.66:41457/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /yolov4_publisher/LRchecktresh: 5
 * /yolov4_publisher/camera_param_uri: package://depthai...
 * /yolov4_publisher/confidence: 200
 * /yolov4_publisher/monoResolution: 400p
 * /yolov4_publisher/nnName: 
 * /yolov4_publisher/resourceBaseFolder: /home/nvidia/catk...
 * /yolov4_publisher/subpixel: True
 * /yolov4_publisher/sync_nn: True
 * /yolov4_publisher/tf_prefix: oak

NODES
  /
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    yolov4_publisher (depthai_examples/yolov4_spatial_node)

auto-starting new master
process[master]: started with pid [30167]
ROS_MASTER_URI=http://192.168.1.66:11311/

setting /run_id to aa7cf384-dbf0-11ec-8a96-48b02d2f7086
process[rosout-1]: started with pid [30180]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [30191]
process[yolov4_publisher-3]: started with pid [30198]
terminate called after throwing an instance of 'std::__ios_failure'
  what():  basic_filebuf::underflow error reading the file: iostream error
Stack trace (most recent call last):
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f9ffc04f8, in raise
Aborted (Signal sent by tkill() 30198 1000)
================================================================================REQUIRED process [yolov4_publisher-3] has died!
process has died [pid 30198, exit code -6, cmd /home/nvidia/catkin_ws/devel/lib/depthai_examples/yolov4_spatial_node __name:=yolov4_publisher __log:=/home/nvidia/.ros/log/aa7cf384-dbf0-11ec-8a96-48b02d2f7086/yolov4_publisher-3.log].
log file: /home/nvidia/.ros/log/aa7cf384-dbf0-11ec-8a96-48b02d2f7086/yolov4_publisher-3*.log
Initiating shutdown!
================================================================================
[yolov4_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

saching13, 2022-05-25:
But stereo examples work ?

iam-shanmukha, 2022-05-25:
> But stereo examples work ?

Yes. Stereo Examples working fine. Getting point cloud also.

saching13, 2022-05-25:
got it. Can you run with `DEPTHAI_LEVEL=debug` and share the output logs ?
ex:
`DEPTHAI_LEVEL=debug roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-POE`

iam-shanmukha, 2022-05-25:
Sure. I will provide the logs tomorrow. Thanks for the follow-up

FPSychotic, 2022-05-25:
https://github.com/luxonis/depthai-ros-examples/issues/46#issue-1237420759

Is the same than this, just the bug was merged in main. Probably because you are using 400p in mono.

FPSychotic, 2022-05-25:
I just tried the main branch, in a different RPi4, and it crash just like before in the devel branch before merge it.
With NN deactivated worked one of five, with NN activated wont work in one, always crash.
It is set at 720p just as default.

iam-shanmukha, 2022-05-26:
> 

- With both 400p and 720p, it is crashing
## 400p
```
DEPTHAI_LEVEL=debug roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-POE
... logging to /home/nvidia/.ros/log/c0271a84-dca9-11ec-aed7-48b02d2f7086/roslaunch-forOak-10865.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.66:45655/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /yolov4_publisher/LRchecktresh: 5
 * /yolov4_publisher/camera_param_uri: package://depthai...
 * /yolov4_publisher/confidence: 200
 * /yolov4_publisher/monoResolution: 400p
 * /yolov4_publisher/nnName: 
 * /yolov4_publisher/resourceBaseFolder: /home/nvidia/catk...
 * /yolov4_publisher/subpixel: True
 * /yolov4_publisher/sync_nn: True
 * /yolov4_publisher/tf_prefix: oak

NODES
  /
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    yolov4_publisher (depthai_examples/yolov4_spatial_node)

auto-starting new master
process[master]: started with pid [10878]
ROS_MASTER_URI=http://192.168.1.66:11311/

setting /run_id to c0271a84-dca9-11ec-aed7-48b02d2f7086
process[rosout-1]: started with pid [10891]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [10894]
process[yolov4_publisher-3]: started with pid [10895]
[2022-05-26 09:40:23.239] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-24 07:25:59 +0000
[2022-05-26 09:40:23.242] [debug] Initialize - finished
terminate called after throwing an instance of 'std::__ios_failure'
  what():  basic_filebuf::underflow error reading the file: iostream error
Stack trace (most recent call last):
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb36ee4f8, in raise
Aborted (Signal sent by tkill() 10895 1000)
================================================================================REQUIRED process [yolov4_publisher-3] has died!
process has died [pid 10895, exit code -6, cmd /home/nvidia/catkin_ws/devel/lib/depthai_examples/yolov4_spatial_node __name:=yolov4_publisher __log:=/home/nvidia/.ros/log/c0271a84-dca9-11ec-aed7-48b02d2f7086/yolov4_publisher-3.log].
log file: /home/nvidia/.ros/log/c0271a84-dca9-11ec-aed7-48b02d2f7086/yolov4_publisher-3*.log
Initiating shutdown!
================================================================================
[yolov4_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
## 720p 

```
DEPTHAI_LEVEL=debug roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-POE 
... logging to /home/nvidia/.ros/log/ddb30c38-dcaa-11ec-8277-48b02d2f7086/roslaunch-forOak-11573.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.66:36421/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /yolov4_publisher/LRchecktresh: 5
 * /yolov4_publisher/camera_param_uri: package://depthai...
 * /yolov4_publisher/confidence: 200
 * /yolov4_publisher/monoResolution: 720p
 * /yolov4_publisher/nnName: 
 * /yolov4_publisher/resourceBaseFolder: /home/nvidia/catk...
 * /yolov4_publisher/subpixel: True
 * /yolov4_publisher/sync_nn: True
 * /yolov4_publisher/tf_prefix: oak

NODES
  /
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    yolov4_publisher (depthai_examples/yolov4_spatial_node)

auto-starting new master
process[master]: started with pid [11586]
ROS_MASTER_URI=http://192.168.1.66:11311/

setting /run_id to ddb30c38-dcaa-11ec-8277-48b02d2f7086
process[rosout-1]: started with pid [11600]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [11614]
process[yolov4_publisher-3]: started with pid [11619]
[2022-05-26 09:48:20.261] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-24 07:25:59 +0000
[2022-05-26 09:48:20.263] [debug] Initialize - finished
terminate called after throwing an instance of 'std::__ios_failure'
  what():  basic_filebuf::underflow error reading the file: iostream error
Stack trace (most recent call last):
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f977384f8, in raise
Aborted (Signal sent by tkill() 11619 1000)
================================================================================REQUIRED process [yolov4_publisher-3] has died!
process has died [pid 11619, exit code -6, cmd /home/nvidia/catkin_ws/devel/lib/depthai_examples/yolov4_spatial_node __name:=yolov4_publisher __log:=/home/nvidia/.ros/log/ddb30c38-dcaa-11ec-8277-48b02d2f7086/yolov4_publisher-3.log].
log file: /home/nvidia/.ros/log/ddb30c38-dcaa-11ec-8277-48b02d2f7086/yolov4_publisher-3*.log
Initiating shutdown!
================================================================================
[yolov4_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

iam-shanmukha, 2022-05-26:
# Update - Seems working after specifying .blob file in lauch file. 

#### yolov4_publisher.launch
```
<arg name="nnName"                default="yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob"         />
```
#### mobile_publisher.launch
```
<arg name="nnName"              default="mobilenet-ssd_openvino_2021.2_6shave.blob"         />

```
## OUTPUT
```
DEPTHAI_LEVEL=debug roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-POE 
... logging to /home/nvidia/.ros/log/e17f480c-dcb1-11ec-9579-48b02d2f7086/roslaunch-forOak-23874.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.66:33351/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /yolov4_publisher/LRchecktresh: 5
 * /yolov4_publisher/camera_param_uri: package://depthai...
 * /yolov4_publisher/confidence: 200
 * /yolov4_publisher/monoResolution: 720p
 * /yolov4_publisher/nnName: yolov4_tiny_coco_...
 * /yolov4_publisher/resourceBaseFolder: /home/nvidia/catk...
 * /yolov4_publisher/subpixel: True
 * /yolov4_publisher/sync_nn: True
 * /yolov4_publisher/tf_prefix: oak

NODES
  /
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    yolov4_publisher (depthai_examples/yolov4_spatial_node)

auto-starting new master
process[master]: started with pid [23887]
ROS_MASTER_URI=http://192.168.1.66:11311/

setting /run_id to e17f480c-dcb1-11ec-9579-48b02d2f7086
process[rosout-1]: started with pid [23900]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [23912]
process[yolov4_publisher-3]: started with pid [23918]
process[rviz-4]: started with pid [23925]
[2022-05-26 10:38:33.026] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-24 07:25:59 +0000
[2022-05-26 10:38:33.028] [debug] Initialize - finished
[2022-05-26 10:38:33.200] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1.tar.xz' open: 5ms, archive read: 167ms
QXcbConnection: XCB error: 148 (Unknown), sequence: 173, resource id: 0, major code: 140 (Unknown), minor code: 20
[2022-05-26 10:38:33.927] [debug] Resources - Archive 'depthai-device-fwp-db9c1703c929df0de6cc99b198d2d149f1735598.tar.xz' open: 5ms, archive read: 894ms
[2022-05-26 10:38:34.345] [debug] Device - OpenVINO version: 2021.4
[2022-05-26 10:38:34.348] [debug] Device - BoardConfig: {"gpio":[],"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 09 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be
[2022-05-26 10:38:35.348] [debug] Booting FW with Bootloader. Version 0.0.16, Time taken: 841ms
[2022-05-26 10:38:35.348] [debug] DeviceBootloader about to be closed...
[2022-05-26 10:38:35.349] [debug] XLinkResetRemote of linkId: (0)
[2022-05-26 10:38:35.888] [debug] DeviceBootloader closed, 539
[10.42.0.37] [172.741] [system] [info] Memory Usage - DDR: 0.12 / 341.19 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.83 / 78.13 MiB, LeonRT Heap: 2.88 / 41.43 MiB
[10.42.0.37] [172.741] [system] [info] Temperatures - Average: 40.22 °C, CSS: 41.66 °C, MSS 39.58 °C, UPA: 40.04 °C, DSS: 39.58 °C
[10.42.0.37] [172.741] [system] [info] Cpu Usage - LeonOS 6.21%, LeonRT: 0.59%
[... 310 lines cut ...]
[10.42.0.37] [270.930] [system] [info] Temperatures - Average: 50.64 °C, CSS: 51.58 °C, MSS 50.25 °C, UPA: 51.58 °C, DSS: 49.14 °C
[10.42.0.37] [270.930] [system] [info] Cpu Usage - LeonOS 99.98%, LeonRT: 24.46%
[10.42.0.37] [271.931] [system] [info] Memory Usage - DDR: 93.28 / 341.19 MiB, CMX: 2.25 / 2.50 MiB, LeonOS Heap: 64.39 / 78.13 MiB, LeonRT Heap: 5.33 / 41.43 MiB
[10.42.0.37] [271.931] [system] [info] Temperatures - Average: 50.64 °C, CSS: 50.92 °C, MSS 50.25 °C, UPA: 51.36 °C, DSS: 50.03 °C
[10.42.0.37] [271.931] [system] [info] Cpu Usage - LeonOS 98.46%, LeonRT: 24.12%
[10.42.0.37] [272.932] [system] [info] Memory Usage - DDR: 93.28 / 341.19 MiB, CMX: 2.25 / 2.50 MiB, LeonOS Heap: 64.39 / 78.13 MiB, LeonRT Heap: 5.33 / 41.43 MiB
[10.42.0.37] [272.932] [system] [info] Temperatures - Average: 50.59 °C, CSS: 50.70 °C, MSS 50.25 °C, UPA: 50.92 °C, DSS: 50.47 °C
[10.42.0.37] [272.932] [system] [info] Cpu Usage - LeonOS 98.65%, LeonRT: 23.76%
[10.42.0.37] [273.933] [system] [info] Memory Usage - DDR: 93.28 / 341.19 MiB, CMX: 2.25 / 2.50 MiB, LeonOS Heap: 64.39 / 78.13 MiB, LeonRT Heap: 5.33 / 41.43 MiB
[10.42.0.37] [273.933] [system] [info] Temperatures - Average: 50.81 °C, CSS: 50.92 °C, MSS 50.92 °C, UPA: 51.80 °C, DSS: 49.59 °C
[10.42.0.37] [273.933] [system] [info] Cpu Usage - LeonOS 95.06%, LeonRT: 28.92%
^C[rviz-4] killing on exit
[yolov4_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
Stack trace (most recent call last):
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f9ee724f8, in raise
Aborted (Signal sent by tkill() 23918 1000)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

FPSychotic, 2022-05-26:
I tried again, everything updated, fresh dai-core, ros and examples. tried specify the file as above, tried different settings as no rviz,lowering resolution to 400p, confidence, still doesn't work. External power supply 4A

`:~$ roslaunch depthai_examples stereo_inertial_node.launch 
... logging to /home/ubuntu/.ros/log/2fd9bb66-dce1-11ec-8bd2-69b5f1bce5e3/roslaunch-ubuntu-44865.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:38017/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 100
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 400p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: yolov4_tiny_coco_...
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/ubuntu/dept...
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://localhost:11311

process[oak_state_publisher-1]: started with pid [44887]
process[stereo_inertial_publisher-2]: started with pid [44888]
process[nodelet_manager-3]: started with pid [44889]
process[depth_image_convertion_nodelet-4]: started with pid [44891]
process[depth_image_to_rgb_pointcloud-5]: started with pid [44896]
[ INFO] [1653563855.094429211]: Initializing nodelet with 4 worker threads.
 nnPath ,, /home/ubuntu/depthai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
Listing available devices...
Device Mx ID: 14442C10517F63D700
[14442C10517F63D700] [34.197] [SpatialDetectionNetwork(8)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
terminate called after throwing an instance of 'std::runtime_error'
  what():  ColorCamera(6) - 'preview' width or height (416, 416) bigger than sensor resolution (640, 360)
Stack trace (most recent call last):
#18   Object "/home/ubuntu/depthai_ws/devel/lib/depthai_examples/stereo_inertial_node", at 0xaaaae31c11a3, in 
#17   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffaa9bae0f, in __libc_start_main
#16   Object "/home/ubuntu/depthai_ws/devel/lib/depthai_examples/stereo_inertial_node", at 0xaaaae31c3847, in 
[... 6 lines cut ...]
#9    Object "/home/ubuntu/depthai_ws/devel/lib/depthai_examples/stereo_inertial_node", at 0xaaaae31ef10b, in 
#8    Object "/home/ubuntu/depthai_ws/devel/lib/depthai_examples/stereo_inertial_node", at 0xaaaae31f28a7, in 
#7    Object "/usr/local/lib/libdepthai-core.so", at 0xffffab355edb, in dai::Device::Device(dai::Pipeline const&)
#6    Object "/usr/local/lib/libdepthai-core.so", at 0xffffab368873, in dai::DeviceBase::tryStartPipeline(dai::Pipeline const&)
#5    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffaabd05df, in __cxa_rethrow
#4    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffaabd027f, in std::terminate()
#3    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffaabd021b, in 
#2    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffaabd28cb, in __gnu_cxx::__verbose_terminate_handler()
#1    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffaa9baaab, in abort
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffaa9cdd78, in gsignal
Aborted (Signal sent by tkill() 44888 1000)
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 44888, exit code -6, cmd /home/ubuntu/depthai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/2fd9bb66-dce1-11ec-8bd2-69b5f1bce5e3/stereo_inertial_publisher-2.log].
log file: /home/ubuntu/.ros/log/2fd9bb66-dce1-11ec-8bd2-69b5f1bce5e3/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[stereo_inertial_publisher-2] killing on exit
[nodelet_manager-3] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

FPSychotic, 2022-05-26:
Yolo publisher works indicating the blob file name as said @[iam-shanmukha

iam-shanmukha, 2022-05-26:
Hi @FPSychotic, I tried changing nnName in `stereo_inertial_node.launch` to `yolotiny...`. But it seems working to me. Which platform you are using?

```
roslaunch depthai_examples stereo_inertial_node.launch
... logging to /home/nvidia/.ros/log/9b08ab32-dcea-11ec-8d36-48b02d2f7086/roslaunch-forOak-671.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.66:40711/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: yolov4_tiny_coco_...
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/nvidia/catk...
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

auto-starting new master
process[master]: started with pid [684]
ROS_MASTER_URI=http://192.168.1.66:11311/

setting /run_id to 9b08ab32-dcea-11ec-8d36-48b02d2f7086
process[rosout-1]: started with pid [708]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [727]
process[stereo_inertial_publisher-3]: started with pid [729]
process[nodelet_manager-4]: started with pid [731]
process[depth_image_convertion_nodelet-5]: started with pid [741]
process[depth_image_to_rgb_pointcloud-6]: started with pid [748]
process[rviz-7]: started with pid [768]
[ INFO] [1653566075.873730938]: Initializing nodelet with 6 worker threads.
 nnPath ,, /home/nvidia/catkin_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
QXcbConnection: XCB error: 148 (Unknown), sequence: 173, resource id: 0, major code: 140 (Unknown), minor code: 20
[... 111 lines cut ...]
[ WARN] [1653566135.276050388]: Accel data not found. Dropping data
[ WARN] [1653566135.550963415]: Accel data not found. Dropping data
[ WARN] [1653566136.317705466]: Accel data not found. Dropping data
[ WARN] [1653566137.257838054]: Accel data not found. Dropping data
[ WARN] [1653566138.927750911]: Accel data not found. Dropping data
[ WARN] [1653566140.254024185]: Accel data not found. Dropping data
[ WARN] [1653566141.009953444]: Accel data not found. Dropping data
[ WARN] [1653566141.868966187]: Accel data not found. Dropping data
[ WARN] [1653566141.956092290]: Accel data not found. Dropping data
[ WARN] [1653566141.957632299]: Accel data not found. Dropping data
[ WARN] [1653566142.802487241]: Accel data not found. Dropping data
[ WARN] [1653566142.987084342]: Accel data not found. Dropping data
^C[rviz-7] killing on exit
[depth_image_to_rgb_pointcloud-6] killing on exit
[stereo_inertial_publisher-3] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[oak_state_publisher-2] killing on exit
[ WARN] [1653566145.305531383]: Accel data not found. Dropping data
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

FPSychotic, 2022-05-26:
I tried the same .launch configuration that @iam-shanmukha posted, 400p,720p , and removed the (find function and written the gull address, still crashes. 
Only difference apart of hardware I can note, is he is in Melodic I'm in Noetic.

FPSychotic, 2022-05-26:
with mobilenet and I activated rviz this time

`~$ roslaunch depthai_examples stereo_inertial_node.launch 
... logging to /home/ubuntu/.ros/log/3fada3b6-dced-11ec-8bd2-69b5f1bce5e3/roslaunch-ubuntu-46168.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:36107/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 100
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: mobilenet-ssd_ope...
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/ubuntu/dept...
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://localhost:11311

process[oak_state_publisher-1]: started with pid [46190]
process[stereo_inertial_publisher-2]: started with pid [46191]
process[nodelet_manager-3]: started with pid [46192]
process[depth_image_convertion_nodelet-4]: started with pid [46197]
process[depth_image_to_rgb_pointcloud-5]: started with pid [46199]
process[rviz-6]: started with pid [46201]
[ INFO] [1653567693.814253802]: Initializing nodelet with 4 worker threads.
 nnPath ,, /home/ubuntu/depthai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/mobilenet-ssd_openvino_2021.2_6shave.blob
Listing available devices...
Device Mx ID: 14442C10517F63D700
[14442C10517F63D700] [418.921] [SpatialDetectionNetwork(8)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
Device USB status: SUPER
[14442C10517F63D700] [419.120] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [419.165] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [419.166] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[... 31 lines cut ...]
[14442C10517F63D700] [419.819] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[14442C10517F63D700] [419.859] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [419.859] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[14442C10517F63D700] [419.900] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [419.901] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[14442C10517F63D700] [419.939] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [419.940] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[14442C10517F63D700] [419.979] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [419.980] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[14442C10517F63D700] [420.019] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[14442C10517F63D700] [420.019] [SpatialDetectionNetwork(8)] [warning] Input image (416x416) does not match NN (300x300)
[14442C10517F63D700] [420.055] [SpatialDetectionNetwork(8)] [error] Mask is not defined for output layer with width '7'. Define at pipeline build time using: 'setAnchorMasks' for 'side7'.
[ WARN] [1653567704.573295324]: Accel data not found. Dropping data
[ WARN] [1653567712.326049576]: Accel data not found. Dropping data
[ WARN] [1653567752.428172847]: Accel data not found. Dropping data
^C[rviz-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[stereo_inertial_publisher-2] killing on exit
[nodelet_manager-3] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

FPSychotic, 2022-05-26:
It worked once in stereo_inertial with same setup than before, as in launch , cables, power supply etc. 
But now don't work again
I think I have a hardware issue with the power system if this camera, like it is weak and the current when is heavy loaded shoot down the camera

FPSychotic, 2022-05-26:
@iam-shanmukha Please, could you try the inertial_stereo example at 400p? It works randomly at 720p, but never works at 400p

FPSychotic, 2022-05-26:
Alright, I could record videos working at 720p, but not at 400p, working and no working at 720p with rviz activated. so if someone gets consistent results at 720p and 400p, it looks crearly I have a hardware issue.

iam-shanmukha, 2022-05-26:
Sure. Understand. I will try at 400p and let you know.

FPSychotic, 2022-05-26:
@iam-shanmukha 
FYI https://github.com/luxonis/depthai-ros/issues/113#issuecomment-1139074295

saching13, 2022-05-27:
Thanks @iam-shanmukha . I will check on theogs and get back. 
No need to test the 400p on stereo internal node. It is a known issue. 

@FPSychotic pleaae don't mix the issues. It will be hard to follow them.

saching13, 2022-05-27:
@iam-shanmukha  I have added the fix on main for yolov4 exampe. it is working now. Please check it out. 
Sorry for the issue.

iam-shanmukha, 2022-05-28:
Thanks @saching13 , the fix is working for yolov4. Could we do the same for mobilenet?

```
roslaunch depthai_examples mobile_publisher.launch camera_model:=OAK-D-POE
... logging to /home/nvidia/.ros/log/cbb5e7dc-de47-11ec-bdd1-48b02d2f7086/roslaunch-forOak-14579.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.66:39625/

SUMMARY
========

PARAMETERS
 * /mobilenet_publisher/camera_param_uri: package://depthai...
 * /mobilenet_publisher/nnName: 
 * /mobilenet_publisher/resourceBaseFolder: /home/nvidia/catk...
 * /mobilenet_publisher/sync_nn: True
 * /mobilenet_publisher/tf_prefix: oak
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    mobilenet_publisher (depthai_examples/mobilenet_node)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [14592]
ROS_MASTER_URI=http://192.168.1.66:11311/

setting /run_id to cbb5e7dc-de47-11ec-bdd1-48b02d2f7086
process[rosout-1]: started with pid [14605]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [14617]
process[mobilenet_publisher-3]: started with pid [14623]
terminate called after throwing an instance of 'std::__ios_failure'
  what():  basic_filebuf::underflow error reading the file: iostream error
Stack trace (most recent call last):
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f9e4064f8, in raise
Aborted (Signal sent by tkill() 14623 1000)
================================================================================REQUIRED process [mobilenet_publisher-3] has died!
process has died [pid 14623, exit code -6, cmd /home/nvidia/catkin_ws/devel/lib/depthai_examples/mobilenet_node __name:=mobilenet_publisher __log:=/home/nvidia/.ros/log/cbb5e7dc-de47-11ec-bdd1-48b02d2f7086/mobilenet_publisher-3.log].
log file: /home/nvidia/.ros/log/cbb5e7dc-de47-11ec-bdd1-48b02d2f7086/mobilenet_publisher-3*.log
Initiating shutdown!
================================================================================
[mobilenet_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

saching13, 2022-05-28:
Yes. Iwill do it Tomorrow.  Thanks for checking.

iam-vishnu, 2022-06-13:
Hi @saching13. I think we can close this as it fixed in cd6b28bc704ed689aa65132b3c5f698455309d78

closed 2022-08-02 by saching13

FPSychotic, 2022-10-11:
I'm not mixing them, you answered about this issue which belong OAK-D in my
other issue that is for OAK-D Pro W Dev and ov9782.
I linked that issue here because you answered with the OAK-D solution
there, instead here.