luxonis/depthai-ros-examples #51: Jetson NX: unable reading color image

bartoszptak, 2022-05-30:
Hi, I'm trying to use ROS nodes for the OAK-D-LITE camera using Jetson NX. Examples from `depthai-python` work correctly, but using `rgb_stereo_node` I get an error.

**Information**
* platform: `Jetson NX`
* os: `# R34 (release), REVISION: 1.1, GCID: 30414990, BOARD: t186ref, EABI: aarch64, DATE: Tue May 17 04:20:55 UTC 2022`
* camera: `OAK-D-LITE`
* ROS: `ros-noetic 1.15.14`
* code version: `main`

**How to reproduce**
* run `roslaunch depthai_examples rgb_stereo_node.launch camera_model:=OAK-D-LITE`
* in second terminal:
  * `rostopic echo /rgb_stereo_publisher/stereo/depth --noarr` -- works well
  * `rostopic echo /rgb_stereo_publisher/color/image --noarr` -- generate error below

**Error**
```bash
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://akl-put:37413/

SUMMARY
========

PARAMETERS
 * /rgb_stereo_publisher/LRchecktresh: 5
 * /rgb_stereo_publisher/camera_param_uri: package://depthai...
 * /rgb_stereo_publisher/confidence: 200
 * /rgb_stereo_publisher/extended: False
 * /rgb_stereo_publisher/lrcheck: True
 * /rgb_stereo_publisher/subpixel: False
 * /rgb_stereo_publisher/tf_prefix: oak
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    rgb_stereo_publisher (depthai_examples/rgb_stereo_node)

auto-starting new master
process[master]: started with pid [9944]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6bceee72-e004-11ec-a249-48b02d2f6e98
process[rosout-1]: started with pid [9955]
started core service [/rosout]
process[rgb_stereo_publisher-2]: started with pid [9962]
Stack trace (most recent call last) in thread 9990:
#25   Object "/usr/lib/aarch64-linux-gnu/ld-2.31.so", at 0xffffffffffffffff, in 
#24   Source "../sysdeps/unix/sysv/linux/aarch64/clone.S", line 78, in thread_start [0xffff9c51722b]
#23   Source "/build/glibc-70LW9c/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0xffff9dd0251b]
#22   Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28", at 0xffff9c6a9fab, in 
#21   Source "/usr/include/c++/9/thread", line 195, in _M_run [0xffff9cef7ec3]
        192: 	{ }
        193: 
        194: 	void
      > 195: 	_M_run() { _M_func(); }
        196:       };
        197: 
        198:     void
#20   Source "/usr/include/c++/9/thread", line 251, in operator() [0xffff9cef7f47]
        248: 	{
        249: 	  using _Indices
        250: 	    = typename _Build_index_tuple<tuple_size<_Tuple>::value>::__type;
      > 251: 	  return _M_invoke(_Indices());
        252: 	}
        253:       };
#19   Source "/usr/include/c++/9/thread", line 244, in _M_invoke<0> [0xffff9cef7fbb]
[... 124 lines cut ...]
         97:     }
#2    Source "/usr/include/c++/9/bits/invoke.h", line 73, in __invoke_impl<void, void (dai::ros::ImageConverter::*&)(std::shared_ptr<dai::ImgFrame>, std::deque<sensor_msgs::Image_<std::allocator<void> > >&), dai::ros::ImageConverter*&, std::shared_ptr<dai::ImgFrame>, std::deque<sensor_msgs::Image_<std::allocator<void> >, std::allocator<sensor_msgs::Image_<std::allocator<void> > > >&> [0xaaaace6914f7]
         70:     __invoke_impl(__invoke_memfun_deref, _MemFun&& __f, _Tp&& __t,
         71: 		  _Args&&... __args)
         72:     {
      >  73:       return ((*std::forward<_Tp>(__t)).*__f)(std::forward<_Args>(__args)...);
         74:     }
         75: 
         76:   template<typename _Res, typename _MemPtr, typename _Tp>
#1    Object "/home/put/catkin_ws/devel/.private/depthai_bridge/lib/libdepthai_bridge.so", at 0xffff9e39a193, in dai::ros::ImageConverter::toRosMsg(std::shared_ptr<dai::ImgFrame>, std::deque<sensor_msgs::Image_<std::allocator<void> >, std::allocator<sensor_msgs::Image_<std::allocator<void> > > >&)
#0    Object "/usr/lib/aarch64-linux-gnu/libopencv_optflow.so.4.2.0", at 0xffff9deee0ac, in cv::Mat::Mat(cv::Size_<int>, int, void*, unsigned long)
Segmentation fault (Address not mapped to object [0x65400000780])
================================================================================REQUIRED process [rgb_stereo_publisher-2] has died!
process has died [pid 9962, exit code -11, cmd /home/put/catkin_ws/devel/lib/depthai_examples/rgb_stereo_node __name:=rgb_stereo_publisher __log:=/home/put/.ros/log/6bceee72-e004-11ec-a249-48b02d2f6e98/rgb_stereo_publisher-2.log].
log file: /home/put/.ros/log/6bceee72-e004-11ec-a249-48b02d2f6e98/rgb_stereo_publisher-2*.log
Initiating shutdown!
================================================================================
[rgb_stereo_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

saching13, 2022-08-10:
Did you try using stereo_inertial_node ?

danilo-pejovic, 2023-05-23:
As there is no activity on this issue I will close it for now. Feel free to let me know if it is still relevant and we can move it to depthai-ros reo.

closed 2023-05-23 by danilo-pejovic