luxonis/depthai-ros-examples #53: OAK-D POE - Setting ColorCameraProperties Sensor Resolution to 12MP

David9696, 2022-06-20:
Hi,

I have a OAK-D POE camera. I would like to test the color image output with 12mp resolution.

This is what I have done:

In stereo_inertial_publisher.cpp, I have set 

camRgb->setResolution(dai::ColorCameraProperties::SensorResolution::THE_12_MP)

However, I got the following error:

OpenCV Error: Assertion failed (sz.width % 2 == 0 && sz.height % 3 == 0 && depth == CV_8U) in cvtColor, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp, line 9878
[2022-06-20 13:49:33.505] [error] Callback with id: 1 throwed an exception: /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp:9878: error: (-215) sz.width % 2 == 0 && sz.height % 3 == 0 && depth == CV_8U in function cvtColor

THE_4_K is working fine. May I know what is the Problem?

Thanks.

saching13, 2022-06-21:
I think this is a decoding issue due to res. Will check and get back.

David9696, 2022-07-20:
Hi, 

Any updates for this? I would like to set the resolution to 12MP.

Luxonis-Brandon, 2022-07-20:
Sorry it looks like we forgot to circle back.  Asking internally.

saching13, 2022-07-21:
Sorry. Forgot to update you.  
 there was a scaling issue for depth align with 12MP. the fix needed changes in the fw. 
  Which is supposedly fixed on the new version. Checking

saching13, 2022-07-22:
@David9696 
When 12MP full resolution is used depth doesn't work on full resolution and arbitary scaled since the hardware needs it to be divisible by 16. So The one of those that works is 8, 39 ratio. In branch `12MP_scaled` I have hardcoded it for you to test. 
If you can run it and check with `rqt_image_view` and confirm. I will change those to ros params and modify the camera info's and create a PR.

[image]

David9696, 2022-07-26:
Hi @saching13 ,

Yes, the  branch 12MP_scaled works well. Thanks. You may merge into main. I would close this issue. Thanks for your great support.

closed 2022-07-26 by David9696