luxonis/depthai-ros-examples #55: feat req: Add left camera and right camera raw color image topics

iam-shanmukha, 2022-07-07:
- Currently getting `/stereo_publisher/left/image` and `/stereo_publisher/right/image` with `roslaunch depthai_examples stereo_node.launch. but the images are greyscale.
- So, Requesting to add right camera and left camera raw color images

saching13, 2022-08-02:
@iam-shanmukha  Since all of our Packaged products are grayscaled on left and right. the stereo_node is configured that way. 
We will be adding a more complex example to handle the custom boards soon.

danilo-pejovic, 2023-05-23:
Sorry for random ping almost year later. We are repurposing this repo for non-cricital ROS examples we develop over time. Like it is said in previous comment our packaged products are grayscaled on stereo cameras.

closed 2023-05-23 by danilo-pejovic

FPSychotic, 2023-05-23:
But FP4 can have 2xRGB cameras , how then setup a FP4 Modular camera with
RGB stereo, that cameras are to precisely can do that, same with OAK-D pro
W Dev.

saching13, 2023-05-23:
@FPSychotic  it should be possible with the new driver in depthai-ros. Please go over the readme there and post an issue if that doesn't help.