luxonis/depthai-ros-examples #57: OAK-D Pro W POE - Launch

David9696, 2022-07-20:
Hi,

May I know which launch file to use for OAK-D Pro W POE?

I tried to launch by using the "stereo_inertial_node.launch", but I got the following errors:

terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

Do I need to change any param?

FYI, I have tried to run the examples in depthai-python. The examples are working.

Luxonis-Brandon, 2022-07-20:
Sorry about the trouble.  Will ask and circle back.

saching13, 2022-07-20:
@David9696  which version of depthai-core (C++) is installed ?

David9696, 2022-07-21:
Hi,

I didn't install the depthai-core explicitly.

I just use the following script:

sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash

I think should be v2.17.1

saching13, 2022-07-21:
got it Thanks. 
Can you try `roslaunch depthai_examples stereo_inertial_node.launch mxId:=<Device ID>` and share the error ?

hamzaMahdi, 2022-07-22:
Setting the mxId manually yields the same error message.

David9696, 2022-07-22:
Hi @saching13 ,

I got the same error as follow:

terminate called after throwing an instance of 'std::runtime_error'
what(): Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

Separately, I have tried this ros package with my oak-d poe. It is working fine. I think this package could not supporting oak-d-pro w poe.

Luxonis-Brandon, 2022-07-22:
Thanks and sorry for the trouble.  Checking.

saching13, 2022-07-22:
> Separately, I have tried this ros package with my oak-d poe. It is working fine. I think this package could not supporting oak-d-pro w poe.

Interesting. Thanks for the info.

saching13, 2022-07-22:
Can you also try exporting the following environment variables ?
```
DEPTHAI_WATCHDOG_INITIAL_DELAY=40000
DEPTHAI_BOOTUP_TIMEOUT=40000
```

hamzaMahdi, 2022-07-22:
I tried setting the environment variable but it resulted in the same output in addition to throwing a warning about the new watchdog timeout

saching13, 2022-07-22:
Thanks alot for testing. Really Sorry for the issue. Will get back with more debug.

In the mean time can you change those above environment variables to `60000` and try it ? 
And also add `DEPTHAI_LEVEL=debug roslaunch depthai_examples stereo_inertial_node.launch` and share the logs please ?

Luxonis-Brandon, 2022-07-22:
Also would you mind shooting an email to support@luxonis.com?  We'd like to send you equipment that should help to debug what is going wrong here.

David9696, 2022-07-23:
> DEPTHAI_LEVEL=debug

Hi @saching13 

I have gotten the same error in the log. I have hidden my device id in the log. The required logs are listed as follows:

[2022-07-23 20:11:22.302] [debug] Library information - version: 2.17.1, commit: c455d09d330bb318211d57690fa9188bfd499640 from 2022-07-19 00:24:11 +0300, build: 2022-07-21 09:51:58 +0000
[2022-07-23 20:11:22.303] [debug] Initialize - finished
[2022-07-23 20:11:22.461] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.19.tar.xz' open: 3ms, archive read: 153ms
Listing available devices...
Device Mx ID:******************** 
[2022-07-23 20:11:22.736] [debug] Device - OpenVINO version: 2022.1
[2022-07-23 20:11:22.736] [debug] Environment 'DEPTHAI_WATCHDOG_INITIAL_DELAY' set to '60000'
[2022-07-23 20:11:22.736] [warning] Watchdog initial delay set to 60000ms
[2022-07-23 20:11:22.737] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":60000,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0d b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be 81 60 ea bb 00 bb 00 be
0020: be be be be be
[2022-07-23 20:11:23.046] [debug] Resources - Archive 'depthai-device-fwp-06e791bb9f6cefc968889a7ae7918ed0ff49e897.tar.xz' open: 4ms, archive read: 738ms
[2022-07-23 20:11:23.105] [debug] Environment 'DEPTHAI_BOOTUP_TIMEOUT' set to '60000'
[2022-07-23 20:11:23.105] [debug] Searching for device: DeviceInfo(name=, mxid= , X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS)
[2022-07-23 20:11:23.124] [debug] Connected bootloader version 0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1
[2022-07-23 20:11:23.124] [info] New bootloader version available. Device has: 0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1, available: 0.0.19
[2022-07-23 20:11:23.995] [debug] Booting FW with Bootloader. Version 0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1, Time taken: 871ms
[2022-07-23 20:11:23.995] [debug] DeviceBootloader about to be closed...
[2022-07-23 20:11:23.996] [debug] XLinkResetRemote of linkId: (0)
[2022-07-23 20:11:24.702] [debug] DeviceBootloader closed, 706
[2022-07-23 20:11:24.710] [debug] Searching for device: DeviceInfo(name=, mxid=*************** , X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

David9696, 2022-07-23:
> Also would you mind shooting an email to [support@luxonis.com](mailto:support@luxonis.com)? We'd like to send you equipment that should help to debug what is going wrong here.

Hi @Luxonis-Brandon,

I have sent an email to support. Please check.

Luxonis-Brandon, 2022-07-23:
Thanks.  Will do.

samialperen, 2022-07-25:
@saching13 @Luxonis-Brandon I am having the same problem with oak-d-poe. I can run demo script without any issue, but I try to run it from ros side, it first can see the MXID and lists the id under available devices, but then it kills itself and says 
```terminate called after throwing an instance of 'std::runtime_error'
what(): Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND```

saching13, 2022-07-25:
> @saching13 @Luxonis-Brandon I am having the same problem with oak-d-poe. I can run demo script without any issue, but I try to run it from ros side, it first can see the MXID and lists the id under [...]

Sorry for the issue. Will get back in a while.

samialperen, 2022-07-25:
Sure! I shared a detailed description here too: https://github.com/luxonis/depthai/issues/753#issuecomment-1194227664

saching13, 2022-07-25:
Can you install depthaai-core of version 2.17.0.0 and try ? [ I testeed on OAK-D-Pro-Poe and had issues with 2.17.1.0. but worked with 2.17.0.0

Here are the steps. 
1. clone depthai-core
2. `cd depthai-core`
3. `git checkout v2.17.0`
4. `cmake -H. -Bbuild -DBUILD_SHARED_LIBS=ON`
5. `cmake --build build --target install`
6. go to ros workspace
7. `catkin_make --cmake-args -D depthai_DIR=<PATH TO DEPTHAI-CORE>depthai-core/build/install/lib/cmake/depthai` 
 
 Let me know if this helps.

saching13, 2022-07-25:
OAK-D-W-PoE on 2.17.0.0
[image]

David9696, 2022-07-26:
> Can you install depthaai-core of version 2.17.0.0 and try ? [ I testeed on OAK-D-Pro-Poe and had issues with 2.17.1.0. but worked with 2.17.0.0
>
> Here are the steps.
>
> 1. clone depthai-core
> 2. `cd [...]

Hi @saching13 ,

No it didn't work. Anything else that I can try?

David9696, 2022-07-26:
> OAK-D-W-PoE on 2.17.0.0 [image]

Can you share you launch file for OAK-D-W-PoE?

Also, can you try with OAK-D Pro W POE ?

Thanks.

saching13, 2022-07-26:
> No it didn't work. Anything else that I can try?

Was the error same ? Can you share some details ? 

> Can you share you launch file for OAK-D-W-PoE?

there is no specific different launch file. `roslaunch depthai_examples stereo_inertial_node.launch` should work. 

> Also, can you try with OAK-D Pro W POE ?

I don't have one at hand as of now. But will do. @daxoft  can you check once you receive yours ?

Luxonis-Brandon, 2022-07-27:
We're checking.  Sorry about delay here.

David9696, 2022-07-27:
> terminate called after throwing an instance of 'std::runtime_error'
> what(): Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

Hi @saching13,

Yes, I got the same error as follows:

terminate called after throwing an instance of 'std::runtime_error'
what(): Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

FYI, initially, my oak-d-pro-w poe was not working with depthai_demo.py. After downgraded the depthai to v3.1.0, it is working now.

https://github.com/luxonis/depthai/issues/751#issuecomment-1195171200

David9696, 2022-07-27:
> We're checking. Sorry about delay here.

hi @Luxonis-Brandon,

I have sent an email to  support@luxonis.com. However, up to date, I haven't received any reply from support@luxonis.com .

daxoft, 2022-07-27:
> > Also, can you try with OAK-D Pro W POE ?
> 
> I don't have one at hand as of now. But will do. @daxoft can you check once you receive yours ?
@saching13  Sure, I still didn't get mine, I tried with OAK-D-PRO-PoE-FF (NG9097 R2M2E2) and it worked with depthai 2.17.0.0 (not with 2.17.1.0, like in your case)

Luxonis-Brandon, 2022-07-27:
> > We're checking. Sorry about delay here.
> 
> hi @Luxonis-Brandon,
> 
> I have sent an email to [support@luxonis.com](mailto:support@luxonis.com). However, up to date, I haven't received any reply from [support@luxonis.com](mailto:support@luxonis.com) .

Sorry about the trouble.  I’m asking what’s going on here.  

Could you please shoot me an email at Brandon at Luxonis dot com as well?  That way if for some reason your email is being blocked by the support system’s SPAM filter I can get around it?

cafemoloko, 2022-07-27:
@David9696 - sorry about the delay, I've just replied to your email

David9696, 2022-07-28:
> Sorry about the trouble. I’m asking what’s going on here.
>
> Could you please shoot me an email at Brandon at Luxonis dot com as well? That way if for some reason your email is being blocked by the [...]

Hi @Luxonis-Brandon,

I would keep you in CC loop.

David9696, 2022-07-28:
> @David9696 - sorry about the delay, I've just replied to your email

Hi @cafemoloko ,

Well received.

Luxonis-Brandon, 2022-07-28:
Thanks, both.  And sorry again about all the trouble here @David9696 .  We're reviewing and changing many processes to make the layers of failure here not happen again.

hamzaMahdi, 2022-08-02:
compiling and using 2.17.0 worked.

I had to download the release since the branch doesnt seem to be up on the repository: https://github.com/luxonis/depthai-core/releases/tag/v2.17.0
I downloaded the depthai-core-v2.17.0.tar.gz file and unziped it. 
then:
```
cmake -H. -Bbuild -DBUILD_SHARED_LIBS=ON
cmake --build build --target **install**

```
After, i catkin_make with the custom directory to depthai:
`catkin_make --cmake-args "-D depthai_DIR=<PATH TO DEPTHAI-CORE>depthai-core/build/install/lib/cmake/depthai"`

It's important to note the "" around the cmake argument starting at -D
Thanks!

samialperen, 2022-08-02:
@hamzaMahdi , verified! It works with Oak-D-PoE.

hamzaMahdi, 2022-08-02:
However, if i try with USB C, it does not work. I opened the PoE encolsure and set pins 2,4,5 to up and the rest to down, then ocnnected usb c. Still getting the XLink error

Luxonis-Brandon, 2022-08-03:
Hi @hamzaMahdi ,

Sorry about the problem.  So to summarize, PoE on the PoE unit works, but when trying to use USB instead, it doesn't work?

If so, one thing to keep in mind is some hosts get confused about USB2 over USB3C cables.  And so forcing USB2 mode may be in order.

But on that, are you wanting to use the PoE camera as a USB camera device?  Or are you doing USB just for debugging?

Thanks,
Brandon

hamzaMahdi, 2022-08-03:
You are correct. PoE worked once I reverted the depthai-core to 2.17.0 
I just wanted to plug in the USB for debugging. I was using a USB3 port

Luxonis-Brandon, 2022-08-05:
Thanks.  So in this case it may be just the case of needing for force USB2 connection (as the PoE series only has USB2 internally).  

https://docs.luxonis.com/en/latest/pages/troubleshooting/#forcing-usb2-communication

David9696, 2022-08-05:
I have updated the depthai-core to 2.17.2. The camera is working in ROS. 

However, when displaying pointcloud in rviz, there are some noise(at about 50cm in front of the camera) when the camera is facing a wall(at about 1m away). Is it meant that the calibration is not good?

snwu1996, 2022-11-17:
Recently got a OAK-D W POE. I tried using depthai-core 2.17.0 and `roslaunch depthai_examples stereo_inertial_node.launch` which was able to run but it only published 1 image and then stopped publishing. Sometimes it wouldn't publish any images at all. However, IMU data was still publishing properly. To fix this I upgraded depthai-core to version 2.19.0 and everything seems to work fine.

danilo-pejovic, 2023-05-23:
We will be using this repository to present some use cases for our ROS nodes that we develop over time. Since this issue seems resolved I will be closing it - feel free to let me know if you are still encountering this issue and we can move that conversation to depthai-ros repo.

closed 2023-05-23 by danilo-pejovic